ardyesp/DLO-138 #6: Where can I find board definition file for Arduino 1.8.4

kyberkiller, 2017-08-31:
Can someone put a link to the Arduino board definition file with which this should compile and work with DSO-138 hardware (I drop the additional parts of the extension away).

I don't get it - this is very much a hardware project but the hardware definitions are not in the project..

Trying to compile with Arduino Due settings (that is the only available ARM 32-bit), compile fails:

`WARNING: library arduino-NVM claims to run on (nRF5, nRF52832, nRF51822) architecture(s) and may be incompatible with your current board which runs on (sam) architecture(s).
In file included from /home/xxx/Documents/stuff/scope/DLO-138/DLO-138.ino:4:0:
/home/xxx/Documents/stuff/scope/DLO-138/DLO-138.ino: In function 'void setup()':
global.h:42: error: 'PB15' was not declared in this scope
 #define BTN4    PB15
                 ^
/home/xxx/Documents/stuff/scope/DLO-138/DLO-138.ino:21:25: note: in expansion of macro 'BTN4'
  loadConfig(digitalRead(BTN4) == LOW);
                         ^
/home/xxx/Documents/stuff/scope/DLO-138/capture.ino: In function 'void setSamplingRate(uint8_t)':
capture:26: error: 'Timer2' was not declared in this scope
  Timer2.pause();
  ^
In file included from /home/xxx/Documents/stuff/scope/DLO-138/DLO-138.ino:4:0:
/home/xxx/Documents/stuff/scope/DLO-138/capture.ino: In function 'void setTriggerRising(boolean)':
global.h:23: error: 'PA8' was not declared in this scope
 #define TRIGGER_IN  PA8
                     ^
/home/xxx/Documents/stuff/scope/DLO-138/capture.ino:38:18: note: in expansion of macro 'TRIGGER_IN'
  detachInterrupt(TRIGGER_IN);
                  ^
/home/xxx/Documents/stuff/scope/DLO-138/capture.ino: In function 'void sampleWaves(boolean)':
capture:57: error: 'Timer2' was not declared in this scope
  Timer2.pause();
  ^
/home/xxx/Documents/stuff/scope/DLO-138/capture.ino: In function 'void startScanTimeout(int16_t)':
capture:70: error: 'Timer2' was not declared in this scope
  Timer2.setCount(2);
  ^
/home/xxx/Documents/stuff/scope/DLO-138/capture.ino: In function 'void scanTimeoutISR()':
capture:142: error: 'Timer2' was not declared in this scope
  Timer2.pause();
  ^
In file included from /home/xxx/Documents/stuff/scope/DLO-138/DLO-138.ino:4:0:
/home/xxx/Documents/stuff/scope/DLO-138/encoder.ino: In function 'void readEncoderISR()':
global.h:40: error: 'PB13' was not declared in this scope
 #define ENCODER_A  PB13
                    ^
/home/xxx/Documents/stuff/scope/DLO-138/encoder.ino:35:26: note: in expansion of macro 'ENCODER_A'
  byte aNow = digitalRead(ENCODER_A);
                          ^
global.h:41: error: 'PB14' was not declared in this scope
 #define ENCODER_B  PB14
                    ^
/home/xxx/Documents/stuff/scope/DLO-138/encoder.ino:36:26: note: in expansion of macro 'ENCODER_B'
  byte bNow = digitalRead(ENCODER_B);
                          ^
/home/xxx/Documents/stuff/scope/DLO-138/interface.ino: In function 'void btn4ISR()':
global.h:42: error: 'PB15' was not declared in this scope
 #define BTN4    PB15
                 ^
/home/xxx/Documents/stuff/scope/DLO-138/interface.ino:24:30: note: in expansion of macro 'BTN4'
  if(!pressed && (digitalRead(BTN4) == LOW)) {
                              ^
global.h:42: error: 'PB15' was not declared in this scope
 #define BTN4    PB15
                 ^
/home/xxx/Documents/stuff/scope/DLO-138/interface.ino:33:29: note: in expansion of macro 'BTN4'
  if(pressed && (digitalRead(BTN4) == HIGH)) {
                             ^
/home/xxx/Documents/stuff/scope/DLO-138/io.ino: In function 'void initIO()':
global.h:21: error: 'PA15' was not declared in this scope
 #define BOARD_LED   PA15
                     ^
[... 221 lines cut ...]
  if(EEPROM.init() != EEPROM_OK) {
            ^
zconfig:11: error: 'EEPROM_OK' was not declared in this scope
  if(EEPROM.init() != EEPROM_OK) {
                      ^
/home/xxx/Documents/stuff/scope/DLO-138/zconfig.ino: In function 'void formatSaveConfig()':
zconfig:124: error: 'struct EEPROMClass' has no member named 'format'
  EEPROM.format();
         ^
/home/xxx/Documents/stuff/scope/DLO-138/zconfig.ino: In function 'void saveParameter(uint16_t, uint16_t)':
zconfig:154: error: 'uint16' was not declared in this scope
  uint16 status = EEPROM.write(param, data);
  ^
zconfig:154: error: expected ';' before 'status'
  uint16 status = EEPROM.write(param, data);
         ^
zconfig:155: error: 'status' was not declared in this scope
  if(status != EEPROM_OK) {
     ^
zconfig:155: error: 'EEPROM_OK' was not declared in this scope
  if(status != EEPROM_OK) {
               ^
exit status 1
'PB15' was not declared in this scope
`

zxfr, 2020-01-16:
I also stuck with compilation troubles. The generic STM32F1x has no Timer2 etc. It would help a lot if you share with us board definitions for DSO138 and DSO150, or at least hints how to adjust the generic board def. Thanks in advance!

dgajew, 2020-01-16:
STM32 Boards / Generic STM32F103C; Variant: STM32F103C8

zxfr, 2020-01-18:
> STM32 Boards / Generic STM32F103C; Variant: STM32F103C8

Thank you for the hint, dgajew! It was a good starting point... 

The standard stm32duino does not include the variant. So as I learned it is also required to copy board definitions from
https://github.com/rogerclarkmelbourne/Arduino_STM32
to ~/Arduino/hardware

With Arduino v1.8.0 and the most recent stm32duino it still did not compile until I installed  "Arduino SAM Boards" from the default list of Arduino boards (obviously it installed a missing specific g++ version)

Now I've got it to successfully compile, hopefully with harmless warnings:
/var/folders/j1/tkr8d9cn6czf9xhy0lkwdb1w0000gn/T//ccODeO0I.s:464: Warning: register range not in ascending order

GHGARC, 2021-03-06:
Hi all
After installing the board / variant: STM32 Boards / Generic STM32F103C; Variant: STM32F103C8 and the Arduino SAM Boards 
the software successfully compile, but the binary once downloaded to the DSO138 is still not working.
This is due to some ports (PB3/PB4, PA13/PA14/PA15) that are allocated to JTAG debug in the  STM32F103C8 after reset.
To overcome this issue the afio_cfg_debug_ports(AFIO_DEBUG_NONE) function needs to be replaced by  disableDebugPorts();
Recompile again, download firmware to DSO138 and enjoy!
Thanks to Ardyesp for this great piece of software!!

kittenarmy, 2023-10-29:
> Hi all After installing the board / variant: STM32 Boards / Generic STM32F103C; Variant: STM32F103C8 and the Arduino SAM Boards the software successfully compile, but the binary once downloaded to [...]

I tried this but it still gave me white screen. Which version SAM did you have installed? I tried 1.6.12.