ardyesp/DLO-138 #8: Compatibility to STM32F303

Martin-Schultz, 2018-03-21:
Hi,

I am really interested in the DSO138, but i would like to be able to sample at higher frequencies.
AFAIK the DSO138 can be used up to a frequency of 200KHZ. However The DS212 uses an STM32F303 instead of an STM32F103 and it can be used up to 1MHZ.

Since both MCUs are pin compatible, i thought about getting an DSO138 and replacing the STM32F103 MCU with an STM32F303. So my question is, whether you know anything about firmware compatibility.

Also i would like to use both ADCs interleaved for the same channel, to improve the sampling rate.

I approeciate any hints or further information.

btd5, 2020-11-02:
Have you made any discoveries? I am planing on switching the chip myself.

Martin-Schultz, 2020-11-03:
Sorry, I went with a different Oscilloscope and completely forgot about this Idea.
Still have one here so this could be a great christmas project.

jools-uk, 2021-05-06:
> Sorry, I went with a different Oscilloscope and completely forgot about this Idea.
> Still have one here so this could be a great christmas project.

Hi - What was your alternative? I was also considering doing exactly as you suggested with STM32F303. But if there is a different kit out there.. I'd be interested to know the options