ardyesp/DLO-138 #10: Adapted for mcufriend 2.4 display support, using mcufriend_kbv library

rtek1000, 2018-07-20:
I asked for help for the original library developer, but he paid no attention, but the library developer mcufriend_kvb paid attention, and now is working the display, I would like to share the modification where only the part of the display has been replaced.

Thank you!

[DLO-138_mcufriend.zip](https://github.com/ardyesp/DLO-138/files/2215431/DLO-138_mcufriend.zip)

stevstrong, 2018-07-21:
Well, taking into consideration that you are using a ***free*** software made by someone in his free time, I cannot understand your ungratefulness just because "the original library developer" had not responded timely to your request. Shame.

rtek1000, 2018-07-21:
Do not be sad,

I think there is a difference between a response from a person who says "here is no place for it" and another answer that says "I will help if you collect more data". It is not the response time, but the content of the response.

stevstrong, 2018-07-21:
I think github is not the right place to discuss about the feature, but bugs.
I told you to open topic on stm32duino forum and you did: http://www.stm32duino.com/viewtopic.php?f=15&t=3886
Ok.
But then you just complain than "I do not pay attention"? This is ridiculous.
You should have wait for my answer on the forum, which cannot be instant, I may be busy with other things. I am sorry to disappoint you if you are expecting that everyone **should** reply to your issue at once.

rtek1000, 2018-07-21:
I modified the comment, if you want I can change it again.

Thank you!

closed 2018-07-22 by rtek1000