ardyesp/DLO-138 #11: Blank screen after flashing code compiled in IDE

kevwag, 2018-08-16:
Using Arduino IDE 1.8.5 with the necessary libraries installed: and having git-cloned your code, I tried to flash the board (I had the Generic STM32F10C Series board selected in tools menu) from the IDE but it kept complaining that it couldn't find the DFU and despite playing with some settings I couldn't get this to work (it would have been very helpful if you had detailed in the readme, which board / CPU speed etc to select to get this working).

To get around this issue, I used the sketch menu to export the compiled code and then flashed it the usual way with the ST flash loader. This appeared to work ok to the point that the file transferred and verified ok, but when powering on after the flash (with the JP1+2 jumpers removed) I just get a blank screen and nothing else. The only code I have altered is to comment out the encoder part in Global.h as per the instruction.

If I flash your pre-compiled binary, that works fine, but I was hoping that once I confirmed that the uncompiled code worked, I could make a couple of minor changes (mainly to add a bit of text to the boot screen) and flash my modified version. I noticed that your binaries are 34k in size but for some reason my compiled binary is 45k - two forward slashes don't use 11k so I have no idea what's going on here??

Some help and clarification on these issues would be greatly appreciated.

ArielBaravalle, 2018-08-16:
hello. 
try add this line bellow  the setup{

`afio_cfg_debug_ports(AFIO_DEBUG_NONE);`

```

#include <EEPROM.h>
#include <Adafruit_GFX.h>
#include "src/TFTLib/Adafruit_TFTLCD_8bit_STM32.h"
#include "global.h"
#include "variables.h"

#define FIRMWARE_VERSION	"1.0"

// ------------------------
void setup()	{
// ------------------------
 afio_cfg_debug_ports(AFIO_DEBUG_NONE);
	DBG_INIT(SERIAL_BAUD_RATE);
	DBG_PRINT("Dual channel O Scope with two logic channels, ver: ");
	DBG_PRINTLN(FIRMWARE_VERSION);

	// set digital and analog stuff
	initIO();
	
	// load scope config or factory reset to defaults
	loadConfig(digitalRead(BTN4) == LOW);
	
	// init the IL9341 display
	initDisplay();
}

// ------------------------
void loop()	{
// ------------------------
	controlLoop();
}

```

infernusdomoso, 2019-02-01:
I had the same issue and can confirm adding that line solved the issue and the firmware boots now.

What did that line do exactly and why would it not being included stop the unit from booting?

PepitoSbazzeguti, 2019-02-03:
Hi,
any news about the above question? I have the same problem.

> I noticed that your binaries are 34k in size but for some reason my compiled binary is 45k - two forward slashes don't use 11k so I have no idea what's going on here??

GHGARC, 2021-03-06:
Hi all
After installing the board / variant: STM32 Boards / Generic STM32F103C; Variant: STM32F103C8 and the Arduino SAM Boards 
the software successfully compile, but the binary once downloaded to the DSO138 is still not working.
This is due to some ports (PB3/PB4, PA13/PA14/PA15) that are allocated to JTAG debug in the  STM32F103C8 after reset.
To overcome this issue the afio_cfg_debug_ports(AFIO_DEBUG_NONE) function needs to be replaced by  disableDebugPorts();
Recompile again, download firmware to DSO138 and enjoy!
Thanks to Ardyesp for this great piece of software!!

kevwag, 2022-04-13:
Yes, all working fine now, thanks all :)

closed 2022-04-13 by kevwag

AlivE-git, 2023-06-09:
I still have a white screen. The solutions above don't work. Are there any other solutions?