ardyesp/DLO-138 #14: Please create some installation manual

saikek, 2019-01-14:
First time flashing it's hard to guess what to do, and how to wire programmer.
Can you please reference official PDF `https://jyetech.com/Products/LcdScope/DSO138_oscilloscope_upgrade.pdf` 

at mention that you can use firmware from your `binaries` folder instead of ofiicial firmware using official uploader.

Thank you!

dgajew, 2019-01-15:
I am affraid you cannot use serial method to upload DLO-138 hex code, I used ST LINK programmer and ST LINK Utility software, without soldering JP1 and JP2 pads. Bootloader probably will be destroyed but this is not problem if you use SWD port instead of TLVL.

saikek, 2019-01-16:
I was using the method mentioned in the PDF with: 
* soldering JP1 and JP2
* with connecting TX, RX, GND terminals to my 3.3v programmer
* using official tool to read firmware and upload new `.bin` file without any issues

ardyesp, 2019-01-26:
The readme does mention "Follow the firmware upgrade instructions for DSO-138". I didn't feel it was appropriate to link the document directly.

closed 2019-01-26 by ardyesp