ardyesp/DLO-138 #21: After firmware flash the voltage is not correct anymore

NilsRo, 2020-04-06:
Hi,

I flashed the firmware on my DSO-138 and now the voltage shown is not correct anymore. On the onboard testsignal it shows Vmax 1.50 and Vmin 0.5. I would expect 3.3 and 0 there. Frequency shown is correct. Any ideas? The firmware looks much better than the original one.

DSO-138 main board PN: 109-13800-00C

Thanks, Nils

morganflint, 2020-04-10:
I've just tested the firmware and the measurements are correct in my case. I'm not sure if it affects the measurements but have you adjusted the zero level by selecting the wave Y cursor and long-pressing OK? 
Another possibility is that there are two hardware versions (even with the same PCB):
- First one's manual and schematic: https://jyetech.com/wp-content/uploads/2018/07/dso138-user-manual-13801k-13802k.pdf. Works with original FW 113-13801-040 or older versions
- Second one: https://jyetech.com/wp-content/uploads/2018/07/dso138-user-manual-13803k-13804k.pdf. Works with original FW 113-13801-050 or newer versions
The only difference is R11, 1.5K in the first version and 150 ohms in the second. This change was introduced to increase the gain of the last stage in the input signal circuit. In the first version, only a small portion of the ADC range was used and this resulted in noise in the trace. According to the schematics in the initial page of this FW, 150 ohms is considered.

I've tested this FW with the newer HW version (R11=150). 

BTW, big thanks to ardyesp for this great FW!!

NilsRo, 2020-04-10:
It is the hardware. I installed the original firmware as newest version and it was the same behaviour. So Iooked a little Deeper and changed resistor R11 yesterday night. But I am not using a metal resistor. The readings are 0,3V off the real values so perhaps I have to go to 180 ohms or a better resistor will fix it.

closed 2020-04-10 by NilsRo

KotieSmit, 2023-11-06:
I also have one of the older boards and replaced R11 with a 150ohm resistor. I too have a bit of voltage difference of 0.35v.