ardyesp/DLO-138 #23: Plotter for DLO-138

HummusPrince, 2020-07-12:
First of all I want to thank you devs for this project.
I'm happy to see that it's still maintained :)

Now I found no tool to parse and plot the serial output of the DLO-138.
So I made one: https://github.com/HummusPrince/DLO-138_plotter

Hope that it'll be useful to someone.

SoniX213, 2020-07-28:
I'm glad to see that this is built for windows but I keep having an issue running it on Linux I have changed the serial port it looks for but pyplot keeps throwing an error every time i try to feed it data

HummusPrince, 2020-07-31:
Hi, thank you for reporting. Would you elaborate what kind of error it gives? Which version of python and pyplot you're using?

CM630, 2021-09-10:
Here is another visualization tool:
https://sourceforge.net/p/lazmer/code/HEAD/tree/trunk/controls/examples/DLO-138/Trepto.exe
[image]

J77M, 2024-12-14:
Hi, I had the same issue with the [DLO-138_plotter](https://github.com/HummusPrince/DLO-138_plotter) as @SoniX213 reported, so I've fetched and modified the plotter by @HummusPrince.
available here: https://github.com/J77M/DLO-138_plotter
The serial input is first read into a buffer, and it should work fine both on Windows and Linux (my case, Ubuntu). Also, I added a better command line interface and a feature to compute the FFT of the signal, which I found useful for my work.