ardyesp/DLO-138 #30: Timebase seems wrong

CM630, 2021-01-08:
On DLO-138_switches_1.0.bin from 26 Feb 2017:
When the indication is 10 ms/div it is actually 5 ms/div.
So a pulse with duration 10 ms shall be displayed in a single section.
When I supply 50 Hz to the input of the oscilloscope at 10 ms I shall see 6 periods (12 section* 10 ms = 120 ms= 6 periods).
Instead, I see 3 periods only.

For example: https://github.com/ardyesp/DLO-138/raw/master/pics/pic4.png (probably made on another version of the software) shows 2 ms per division chosen, and about 1 period of 60 Hz visualized, but with the aforementioned version I see about half a period at 50 Hz. 

Also, when a waveform is sent over UART, the value of „Per Sample (us):“ and „Timebase: 10 mS/div“ seem wrong, the Time column- too.
[image]

ardyesp, 2021-08-09:
Hi @CM630 ,
I am not sure why it is showing incorrect on your hardware. You can see other images in the pics directory and the frequency and timebase are all correct. 

Does it happen on 10ms timebase only?

CM630, 2021-08-23:
It happens on all timebases.
Please note:
1. I have tested in a single channel mode (only channel A1 is enabled), I will check what happens if I enable a second channel.
2. I have used _DLO-138_switches_1.0.bin_ from 26 Feb 2017 which I have downloaded from github. I tried to build a new BIN file from the sources, but I did not succeed.

CM630, 2021-08-29:
I have tried with all 4 channels enables - still the timebase is wrong.
I suppose that my device is not a genuine no - ther is no DSO-138 printed on it.
The processor is STM32F 103C8T6 (the "6" rather looks like "ь") 9910X RR93.
I have found some instructions how to build, probably in https://www.stm32duino.com (as it is mentioned in the readme.md), but I did not suceed, so I cannot fix myself.