ardyesp/DLO-138 #31: Download compiled binary

Einstein2150, 2021-07-20:
Can someone give me a compiled download? I compiled mine with no errors, flashed it but there is only a white screen. I've downgraded the library like it is mentioned in the manual.

Flashing the stock firmware install without problems 🤷🏻‍♂️

ArielBaravalle, 2021-07-21:
hello

you try  to  put  this line in the first line os setup?
` afio_cfg_debug_ports(AFIO_DEBUG_NONE);`

`
// ------------------------
void setup()	{
// ------------------------
 afio_cfg_debug_ports(AFIO_DEBUG_NONE);
	DBG_INIT(SERIAL_BAUD_RATE);
	DBG_PRINT("Dual channel O Scope with two logic channels, ver: ");
	DBG_PRINTLN(FIRMWARE_VERSION);

	// set digital and analog stuff
	initIO();
	
	// load scope config or factory reset to defaults
	loadConfig(digitalRead(BTN4) == LOW);
	
	// init the IL9341 display
	initDisplay();
}

`

kittenarmy, 2023-10-29:
I tried flashing the binary too but it didn't work for me.

CM630, 2023-11-06:
„it didn't work for me.“ is not too descriptive.
What happenned after you flashed the device?

kittenarmy, 2023-11-07:
> „it didn't work for me.“ is not too descriptive. What happenned after you flashed the device?

I just got white screen IRRC. From my digging I think my DSO-138 has a different LCD to the expected 9341. I'm waiting for ST Link to arrive to do some dev as I got tired of soldering jumpers for each compile.

Einstein2150, 2023-11-10:
> >... I got tired of soldering jumpers for each compile. 

I soldered 2 breadboard jumper wires on it and bridged them on the breadboard 😅

kittenarmy, 2023-11-10:
> > > ... I got tired of soldering jumpers for each compile.
> 
> I soldered 2 breadboard jumper wires on it and bridged them on the breadboard 😅

Aha I just did the same. :smile: Seems like even basic things like getting the board LED to flash or TFT black screen needs that `afio_cfg_debug_ports(AFIO_DEBUG_NONE);` which renders SWD/STLink useless. Oh well. :disappointed_relieved:

CM630, 2024-01-03:
Did someone succeed to make a working BIN from the source?
The uploaded BIN works for me, but it needs some fixes.