ardyesp/DLO-138 #33: Blank screen

AlivE-git, 2023-06-10:
Hi. I have a white screen issue. "afio_cfg_debug_ports(AFIO_DEBUG_NONE)" and "disableDebugPorts()" don't help. But the factory firmware works fine.
Also, the serial port works fine, so the problem is definitely with the display.
I tried to build the firmware from the sources, as well as precompiled binaries from the repository. 
Perhaps some hardware modification is needed?

kittenarmy, 2023-10-29:
I think I have the same display or variant. Looking at the ribbon cable under the screen it says CL9325-115. TFT board is PN 109-13700-00A1. 

I tried adding `#include "src/TFTLib/ili932x.h"` to DLO-138 and put `tft.begin(0x9325);` in display; I get some upside down images not quite drawn correctly. 

[image]
[image]

kittenarmy, 2023-10-29:
After updating to Adafruit GFX 1.2.7 library I get some waveform too. But still upside down and draw problems. 

[image]

kittenarmy, 2023-11-12:
> So, in my case, the display controller is lii9325. If I change it in the tft.display function it starts working, but it looks like fillScreen is not working properly. I tried the examples taken from the Adafruit_TFTLCD_8bit_STM32 library repository, they also do not work correctly. [image]

Comparing this project code to [Adafruit_TFTLCD_8bit_STM32](https://github.com/stevstrong/Adafruit_TFTLCD_8bit_STM32) there's sections that could be causing some elements to display upside down like in your second picture. Additionally there is some 8bit or 32bit vs originally 16bit variables that could be messing with display.

AlivE-git, 2023-11-12:
I haven't looked into the project for a long time, but the screen has worked for me.
I replaced "tft.begin(0x9341)" with "tft.begin(0x9325)", "tft.setRotation(LANDSCAPE)" with "tft.setRotation(3)", "tft.full Screen(ILI9341_BLACK)" with "tft.fillRect(0, 0, TFT_WIDTH, TFT_HEIGHT, ILI9341_BLACK);".
But the oscilloscope was not working correctly. The values on the display were updated only by pressing the keys. There may be a problem with interrupts.

kittenarmy, 2023-11-14:
I've been poring over the code and it seems like the main problem is fillScreen GFX function isn't working. Everything else is drawing correctly. It's just a mess without fillScreen. But there is a snag, fillScreen function seems to rely on setAddrWindow but for me I can't call it directly or it crashes.

kittenarmy, 2023-12-21:
I replaced fillScreen with fillRect and that helped, but I'm still missing a lot of the size1 text. 
[image]

Not sure why the text isn't working, I can get these functions to work fine in a simple sketch. I see on your pics this small text from banner is displaying, whereas I only get DSO-138 in a box.